MaslowCNC/Maslow-Create #4: UI Refactor

BarbourSmith, 2019-02-01:
The overall interface needs to be cleaned up and made to look nice. 

Todo:

- [ ] Add dividing lines between sections of the UI
- [ ] Make color scheme of JSCAD area match the rest
- [ ] Continue fade throughout
- [ ] Load projects with their images

BarbourSmith, 2019-03-05:
@alzatin This is what we're going to work on first

alzatin, 2019-03-06:
"Make attachment points spawn above and below the atom" 
Do you mean like a wrap around while you are hovering over ? Or so that the attachment points span around the atom when they are not being hovered over?

alzatin, 2019-03-06:
[image]

BarbourSmith, 2019-03-06:
I was imagining like a combination of the lower right option and the left option, but it's really your call!

[image]

alzatin, 2019-03-06:
[image]
[image]

// Top edge and bottom edge//  ? I'm not sure I'm doing the changes in the right places

alzatin, 2019-03-06:
[image]

BarbourSmith, 2019-03-06:
That looks AWESOME!!

By top and bottom edge I mean if the screen. Like if the atom is close to
the tip of the screen (like test with "translate" which has four inputs)
when they expand they can go off the top of the screen so you can't get to
them anymore.

alzatin, 2019-03-07:
I know. That's what's happening in the pictures. It switches direction if it's too close to the top but it's one directional :(  but I think I need to make it spread up and down in relation to the radius unless it's too close to the top or bottom :)

BarbourSmith, 2019-03-07:
Very cool! I didn't notice the top of the screen in the picture.

Making it two sided sounds like the next step for sure.

BarbourSmith, 2019-03-13:
The next big thing on this list is to make it so that everything on the screen resizes based on how much space is available so that if you resize the screen all of the atoms will get smaller and bigger. The goal is that it won't matter what size screen you have, everything will scale to look the same overall. I think the way to do that is to multiply everything by a scale factor that we compute every time the screen is resized.

I'm about to change how all of the files are organized so be prepared for things to get a little messed up

closed 2019-04-30 by BarbourSmith