MaslowCNC/Maslow-Create #53: Delete and Backspace to remove molecules

ajthinking, 2019-03-29:
Attempted fix to allow Mac user to delete Molecules.
[image]

But noticed it is in conflict with typing in the parameter boxes:
[image]
Im assuming it is the same for Delete button but only have Backspace to test with. Is it possible to scope the active element somehow?

BarbourSmith, 2019-03-29:
Good catch. I adding the line `this.parent.backgroundClick()` in the deleteNode() function should fix that issue

[image]

ajthinking, 2019-03-29:
Really? The delete statement is above that? I tested to add it here
[image]

and it seems to work

BarbourSmith, 2019-03-29:
The backgroundClick() method will simulate clicking in the background which means nothing is selected which will update what is shown in the 3D preview and what values are shown in the sideBar

ajthinking, 2019-03-29:
Hmm, OK! Harder than I taught, have to leave now :(
Hope what you did there fixes it. Could not get that to work here for some reason

PR #55 (2019-04-01): Fix using backspace to delete elements
Closes #54
Closes #53
commit deaef9b5 (2019-04-01): Fix using backspace to delete elements

Closes #54
Closes #53
closed 2019-04-01 by BarbourSmith