MaslowCNC/Maslow-Create #99: Disturbing the peace - greetings from Gili

ajthinking, 2019-05-05:
Hi my Dude, how are you? Hows the project coming along?
I just learned a neat little trick today thought might be handy in your project :)

See this list:
[image]

It probably gonna grow right? So you can probably do this instead:
```let molecules = require.context('./molecules', false, /\.js$/);```

to import all js files in that folder 🚀 🚀 🚀

BarbourSmith, 2019-05-06:
OOOOOO that is slick. Very nice!

How do you like the Gilis? I miss Bali a lot, what a place!

ajthinking, 2019-05-07:
Slick is da word! Gilis really nice. No cars no bikes no nothing. just horses :)

closed 2019-05-07 by ajthinking