MaslowCNC/Maslow-Create #158: Issue with check boxes spawning checked always

BarbourSmith, 2019-05-27:
**Describe the bug**
There seems to still be an issue with the check boxes. They work properly until you click away from the 3D view and then click back. When you click back to the 3D view the check boxes load checked regardless of the actual state of what they represent. 

**To Reproduce**
Steps to reproduce the behavior:
1. Click on the 3D preview 
2. Un-check one or more check boxes
3. Click on the flow drawing part of the program and move an atom around or something
4. Click back on the 3D preview to see all the check boxes checked again

**Expected behavior**
The check boxes should load unchecked 

**Screenshots**
[image]

BarbourSmith, 2019-05-27:
@alzatin

alzatin, 2019-05-28:
Heard

alzatin, 2019-06-03:
To add to the issue i'm pretty sure the axes gets buried as soon as any shape is built. I'm not sure if it's possible to make it longer or visible over the material but i'll check

BarbourSmith, 2019-06-03:
I'm pretty comfortable with it getting buried, I think that other CAD programs all have their axis get buried

BarbourSmith, 2019-07-02:
Fixed

closed 2019-07-02 by BarbourSmith