MaslowCNC/Maslow-Create #187: Outline disappears on translate

BarbourSmith, 2019-07-18:
**Describe the bug**
When you change a value in a translate atom, the outline disappears after.

**To Reproduce**
Steps to reproduce the behavior:
1. Click on a translate atom
2. Change one of the values
3. Press "enter"
4. See that the outline has gone away

**Expected behavior**
The outline should stay

**Screenshots**

[image]

alzatin, 2019-09-09:
The last pull request should take care of the color issue when changing the editable values on all the molecules

BarbourSmith, 2019-09-10:
If you add the words "fixes #187" to your pull request this issue will close automatically when that pull request is merged