MaslowCNC/Maslow-Create #192: Make it more clear when a project saves

BarbourSmith, 2019-07-18:
**Is your feature request related to a problem? Please describe.**
I find that when I am designing things, I often open the console just to make sure that they are saving properly. It would be nice if a successful save was indicated somewhere on the screen.

**Describe the solution you'd like**
Maybe if it said "project saved" in the new top bar space, or even better if it had a timer until it next saved.

**Describe alternatives you've considered**
Opening the console to see the console.alert("project saved") happen works, but that won't work for most people.

@alzatin do you think this could be a part of the UI changes?

alzatin, 2019-07-18:
It definitely can.

alzatin, 2019-09-09:
Hey just wanted to check in on this. I saw you put the saving... box while it is saving. If you think saving time is going to be longer at any point I'd love to make the green progress circle we'd talked about or something like a progress bar but it seems like it's only worth it if it takes more than a second or so. LMK.

BarbourSmith, 2019-09-09:
For very complicated projects when my internet connection is slow I'm seeing 40+ seconds to save. I had to move saving into a new thread so it wouldn't lock everything up while it was saving, but now things are pretty good. I think this is pretty good the way it is and we can close this issue.

closed 2019-09-10 by BarbourSmith