MaslowCNC/Maslow-Create #237: Circle Segment

alzatin, 2019-09-09:
**Is your feature request related to a problem? Please describe.**
I think something that is unclear to me is why the CircleSegment input value is part of the global sidebar instead of just the Circle molecule.

**Describe the solution you'd like**
If it just affects the circle could we move it so it just displays in the circle molecule? That way maybe we can use the rest of the sidebar for let's say a dropdown of available molecules?

Maybe i'm totally misguided, sorry if that's the case

BarbourSmith, 2019-09-10:
This is totally a confusing thing that should be fine better. Originally it was an option for each circle, but I wanted a way to be able to adjust it globally for the whole project because it's useful to set a lower value so things run smoothly on my computer and then change it to a higher value when I'm ready to export the files for cutting (which makes my computer really slow).

I agree we need a better solution, but I'm not sure what it is. What do you think?

alzatin, 2019-09-10:
So let me get it straight. When you change the value of circle segment it also adjusts everything that has already been created, right? If that's the case, maybe it's just a matter of language and we can add something that clues you in on why and when you'd want to change that value. I don't know if resolution/definition/number of segments are the right words but maybe if there's a range of values that this number should remain in we can make a sliding bar suggesting that the lower end is for building/testing and higher end is for exporting, or something like that??

BarbourSmith, 2019-09-10:
I like that plan a lot. Right now it doesn't trigger an update, it only updates when the model is reloaded (but that is a bug).

I think a slider is the way to go.

alzatin, 2019-09-13:
Gotcha!

closed 2019-11-22 by BarbourSmith