MaslowCNC/Maslow-Create #338: The properties button is called "intersection"

BarbourSmith, 2020-01-08:
**Describe the bug**
The properties menu button has the image and name "intersection"

**To Reproduce**
Steps to reproduce the behavior:
1. Open the right click menu
2. Hover over properties

**Expected behavior**

It should be called "properties" and have an appropriate picture 

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - Windows
-  Firefox

alzatin, 2020-01-24:
This is fixed, I'm not sure why it didn't close automatically

BarbourSmith, 2020-01-24:
I think you need a space between them like "Fixes #123"

closed 2020-01-24 by BarbourSmith