MaslowCNC/Maslow-Create #339: Everything breaks in tall narrow mode

BarbourSmith, 2020-01-08:
**Describe the bug**

Right now we don't handle weird aspect ratios well at all

**To Reproduce**
Steps to reproduce the behavior:
1. Drag the window to be tall and narrow.
2. Refresh the page and go through the login process. Lots of things don't play nice

**Expected behavior**
Buttons should resize to fit in the new space and the flow canvas should take up as much as possible to maintain it's aspect ratio but not expand to fill half of the space vertically even if it is not used.

**Screenshots**

[image]

[image]

[image]

**Desktop (please complete the following information):**
 - Windows
 - Firefox

alzatin, 2020-01-24:
I guess this is a bigger question than just changing the buttons to resize since how I understand this is usually handled is with media queries but it requires some forethought and decisions regarding what kind of devices we are targeting. I can add media queries for a phone or a tablet aspect ratio but in reality our atom resizing would make the atoms so small that they would not be usable at all.

alzatin, 2020-01-24:
i think if we do want to handle tablets or phones well we probably need to do a layout that is specific for those so it can change the size according to screen size.. maybe?

BarbourSmith, 2020-01-24:
I agree that if we really wanted to target phones and tablets and stuff it is a separate issue. I'm thinking more of how the window looks when I resize it on my computer.

I think what we need to do is declare that the atom size is like 1/40th of the window width always. That works if we force the canvas aspect ratio to always be the same. I think that will fix #348 as well.

BarbourSmith, 2020-01-24:
I just looked at the code for the size pull request and I guess this is basically already done and the ratio is 1/72 😀 

Now I think we just need to force the aspect ratio to be constant so we don't end up with dead space

alzatin, 2020-01-24:
Yeah forcing the ratio of the canvas would be good but then do you want the viewer to get taller or do we just want dead space around our already established ratio?

alzatin, 2020-01-24:
Also i think the 1/72 ratio works in theory if the screen is always loaded in the same pixel ratio but if you load a project in a smaller screen width it loads a size that never makes it to normal size even if you resize the window to be bigger (i added this as a new issue but i'm not sure how to solve this yet)

BarbourSmith, 2020-01-24:
I think we may need to pad dead space to the side if the height is the limiting factor, but I don't think we should be putting dead space below since we have good content to show there. Having the 3D view be larger is always nice.

I guess maybe we need to use an if statement to check if the height or the width is the limiting factor on preserving the aspect ratio and then resize the other accordingly?

alzatin, 2020-01-30:
[image]

Ok so what if we just make it resize like this ? Does this look right? It's not really usable at that scale but it won't have the odd empty space anymore. I guess we could probably give it some padding at the top so it doesn't interfere with the menu

BarbourSmith, 2020-01-31:
That looks fantastic! Totally right. I wouldn't worry about the menu, I think that seems great.

PR #355 (2020-02-22): Aspect ratio issue
fixes #339

I think this should fix it. There might still be some weird issues if the page is started in very narrow mode but i've messed up the placement (due to resizing compression issue) of all of my projects and it seems to be working with the ones that are still in order but let me know how it looks on your side. 
Also i updated and it seems to be throwing me a console error for 'showdown' -missing module. Not sure what I should do.
closed 2020-02-22 by BarbourSmith