MaslowCNC/Maslow-Create #340: We can remove the resize toggle

BarbourSmith, 2020-01-08:
**Is your feature request related to a problem? Please describe.**

I know it was my idea to add it, but at this point we want to start making the features we have work really well and remove the ones we don't need. I think the size toggle switch can be removed since I've never used it. As it exists now without any label it isn't clear what it does and it somewhat confuses the drawing area.

[image]

**Describe the solution you'd like**
Delete it

**Describe alternatives you've considered**
We could keep it and add a label to it, or move it into the side bar where it would make more sense next to the circle segments slider, but I really don't think we need it.

alzatin, 2020-01-24:
i think we don't need it as long at we stick with the smaller size for the atoms because i find myself using it in the projects where there's a lot of atoms. But i think not having it is cleaner and goes more in line with encouraging people to have less atoms on each screen. I think if the user settings thing ends up being implemented, the size of the atoms should be one of the things you can pick but for now i think i'll just delete it

BarbourSmith, 2020-01-24:
That sounds great to me 😃

PR #347 (2020-01-24): Remove switch and resize initial atom fixes #340
fixes #340

Remove atom size switch and resize initial atom to be smaller
commit f37d2be1 (2020-01-24): Merge pull request #347 from alzatin/fixes#340

Remove switch and resize initial atom fixes #340
closed 2020-01-24 by BarbourSmith