MaslowCNC/Maslow-Create #351: Add Copy and Paste Functionality

BarbourSmith, 2020-01-29:
**Is your feature request related to a problem? Please describe.**

I'm frustrated when I create something in one part of a design and want to move it to another part.

**Describe the solution you'd like**
It would be nice if there was a way to select an atom and then do Ctrl + C to make a copy of it and then Ctrl + V to make a paste of it. 

**Describe alternatives you've considered**

I thought about if we want copy and paste on screen buttons, but I think that would make things too cluttered

**Additional context**

I think this shouldn't be too hard. What we need to do is to serialize the molecule when it is copied and then de-serialize it when it is pasted.

alzatin, 2020-02-01:
By de-serialize it you mean once it's placed it gets a new unique ID ?

BarbourSmith, 2020-02-02:
There is a function in atom called "serialize" and another called "deserialize" which basically take the atoms and convert them into text strings which we use to save them. Then later when we want to load them we can convert back from text strings. I wrote the functions for saving and loading projects, but we can use the same functions to do copy and paste too since basically when we copy an atom we need to convert it into a string so we can store it for later, then when we paste it we just convert back from the string.

https://en.wikipedia.org/wiki/Serialization

Don't worry about fixing this issue, I can do this one or we can skype about it and you can fix it down the road if you get to everything else

BarbourSmith, 2020-02-21:
I think the steps we should use to make this happen are something like this. I'm sure I'm wrong and we'll learn why along the way, but at least it's a start.

1) Have a copy() and paste() function which run when you press control+c and control+v. Those functions should probably be in global variables? 

2) Have a function in atom called copySelected() which calls the serialize() function on that atom if it is selected. This will return a json string. At this point let's just print that string out.

3) Save that string to a variable in global variables called "coppied"

4) When the function paste() runs print that string out

5) De-serialize that string to paste it. This is going to require a bit of a re-write of the de-serialization process so let's skype for this part

alzatin, 2020-02-25:
I think i'm there (#5)

BarbourSmith, 2020-02-25:
Let's chat today?

PR #384 (2020-04-19): Multiselect 2
Fixes #351 This MultiSelect option works by clicking CTRL and dragging a box across the atoms. It is working great with copy paste but there's still some issue with the DeleteNode like with the initial multi select. I created a new molecule for the box, i think it worked out pretty good but let me know what you think of this solution.
closed 2020-04-19 by BarbourSmith