MaslowCNC/Maslow-Create #368: Sidebar disappear when resizing

BarbourSmith, 2020-02-22:
**Describe the bug**

When the window is resized, the bar over on the right hand side disappears.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a project
2. Click on an atom to look at it's sidebar
3. Reduce the window size and notice that the sidebar starts to disappear as the window gets smaller 
4. The 3D view will take up the whole space

**Expected behavior**
The ratio of space between those two elements should stay constant

**Screenshots**
[image]

[image]

alzatin, 2020-02-22:
mhm... Let me check if we have the same code because mine is resizing fine.
[image]

[image]

BarbourSmith, 2020-02-22:
🥺...why is mine not working?

BarbourSmith, 2020-02-22:
What do you see on the .org site?

alzatin, 2020-02-22:
it's working
[image]

BarbourSmith, 2020-02-24:
Weird, OK I will take this one since I'm the only one seeing it 😛 

What browser are you using?

alzatin, 2020-02-25:
Chrome. I just checked on Firefox and it's broken there. I can try to fix it if you want now that i can see it

alzatin, 2020-02-25:
The grid looks different too :(

BarbourSmith, 2020-02-25:
I think you understand how the system works better than I do so if you want to fix it now that you can see it go for it!

PR #407 (2020-04-16): Fixes #368
Fixes #368 
Should fix the resizing problem on Firefox 
I would add that one of the issues we need to fix is to add some media queries if the screen gets too small.
commit 18c9a2ea (2020-04-16): Merge pull request #407 from alzatin/fixes-#368

Fixes #368
closed 2020-04-16 by BarbourSmith