MaslowCNC/Maslow-Create #408: Projects Clean Up

alzatin, 2020-04-16:
**Is your feature request related to a problem? Please describe.**
Maybe you'll think this is a terrible idea but I think we should clear out some projects created before everything was working better since there's so many projects that seem kinda useless and unusable.

**Describe the solution you'd like**
Delete or remove from project line-up projects that are unusable because of the stage in the project they were created in .

**Describe alternatives you've considered**
Not do anything to the line up but select a few projects that can be featured as examples so that when someone who has never seen the program tries it for the first time are not confused on why so many projects don't work

**Additional context**

alzatin, 2020-04-16:
or we could do some sort of tag/ color/labeling to warn that the project might not be currently working

BarbourSmith, 2020-04-16:
I see what you mean. There is a lot of junk out there.

I have two thoughts about this. 

First, I intentionally set things up so that only the users have control over their files. A lot of cloud based CAD programs are criticized for things like "What if the company goes out of business, do I loose all my files?" and there are issues like what if someone puts up something copyrighted, can I be held liable for that? The solution is that we are just making a piece of software which lets people interact with their own files in their own github accounts. Google has spent a ton of money establishing precedent that search engines are protected from being liable for any of the stuff which comes up in a search...so we're basically a search engine. Unfortunately, that means we are unable to delete any projects the same way google can't delete websites.

What we can control is the order in which things appear and how they look. Right now the order is done based on how many stars a project has. I am hoping that once more projects are added and people start using the service the good projects will get enough stars to float to the top.

A quick solution could be that we could go through and star all the good projects so they end up at the top of the results? I think at the end of the day there will be thousands of dead end projects no matter what we do so we should focus on calling attention to the good ones, instead of hiding the bad ones

alzatin, 2020-04-19:
I think starring is a good idea but in a way for it to work we can only star projects we either own or have forked because otherwise we don't know if the molecule arrangement is good or not.

BarbourSmith, 2020-04-19:
To be honest at this point I think the only projects which are any good are projects we own 😁 

There was some interest when I announced the project last year, but since then I think the number of people using it other than us has gone to zero

alzatin, 2020-04-27:
The filter search currently i think organized the projects by when they were last updated but maybe we can change it to stars and star a bunch of projects?

BarbourSmith, 2020-04-27:
I think that sounds like a fantastic idea. I think the search when you are placing a github molecule is organized by stars and that's what I was thinking of 😬

closed 2020-07-04 by github-actions[bot]