MaslowCNC/Maslow-Create #410: Tutorial original project needs fixing

alzatin, 2020-04-17:
**Describe the bug**
the initial project that every user receives has molecules with the wrong position (all in the corner)

**To Reproduce**
Steps to reproduce the behavior:
Open the file "Tutorial" that all users get added to their folder

**Expected behavior**
I expect the file to be a working file from which I can learn how to use the program. Original tutorial file needs to be fixed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

BarbourSmith, 2020-04-18:
Does everyone still receive the tutorial project? I thought I decided that was a bad idea and got rid of it

alzatin, 2020-04-19:
Well I only noticed it because I deleted all my projects and once i had 0 projects it reappeared

BarbourSmith, 2020-04-26:
This has been removed

closed 2020-04-26 by BarbourSmith