MaslowCNC/Maslow-Create #411: Problem with Delete

alzatin, 2020-04-17:
**Describe the bug**
Delete key Deleting molecule even when text box is active//
I'm trying to fix it at the moment

alzatin, 2020-04-17:
[image]
**How to Reproduce behavior**
Let's say i have a molecule and I'm trying to change its name, it it's selected it deletes it but if it's not selected and i'm in the text box and click delete it deletes it and gives me this console error message. Any clue you know what it's going on?

BarbourSmith, 2020-04-18:
My guess is that it's trying to save the new value for the deleted atom. I wouldn't worry about it. My guess is that the issue will go away when the delete problem is fixed. 

I think I fixed something similar with copy and paste a while ago, basically you couldn't copy and paste in to the text boxes without it copying and pasting the atoms. Check out line 138 in flowDraw:

`if (GlobalVariables.ctrlDown && e.keyCode == cKey && document.activeElement.id == "mainBody") {`

Basically it checks to see if the active element is the canvas container before doing the copy or paste.

BarbourSmith, 2020-04-26:
This is fixed now, right?

alzatin, 2020-04-27:
Yes the problem with delete in the text input fields is fixed

closed 2020-04-27 by BarbourSmith