MaslowCNC/Maslow-Create #419: You can't create a project with the letter 'C' in the name 😬

BarbourSmith, 2020-04-26:
I think it probably has something to do with copy and paste?

BarbourSmith, 2020-04-26:
You also can't copy or paste into a project title

BarbourSmith, 2020-04-26:
I'm also not able to paste into a code atom. These are all seen while using chrome

alzatin, 2020-04-27:
yeah sounds like it has to do with the event default we put on those keys. mhm .. when i was trying to fix the search bar i realized some content wasn't considered Editable unless you made it editable. Could be something to do with that? Since that's the condition for the event the fault to not be triggered

BarbourSmith, 2020-04-27:
Smart, I bet that is exactly what is going on. Maybe we should be using a different criteria other than editable?

PR #422 (2020-05-01): Fix default text errors
Maybe Fixes #419 
So, I'm not having the problem with copy/paste in the atom title but I fixed the C problem. I don't know if this is kind of a janky solution but the problem with code is that it creates text areas as it gets written and it seemed easy to test to see if it was a test area. We could combine it all into one if statement to prevent the default but having one that turned the content editable seemed more clear.
closed 2020-05-01 by BarbourSmith