MaslowCNC/Maslow-Create #428: Broken delete selection

alzatin, 2020-05-25:
**Describe the bug**
The action that should delete atoms does not delete all selected atoms.

**To Reproduce**
Select multiple atoms and try to delete them with delete or backspace.

**Expected behavior**
All selected atoms should be deleted if this key is pressed

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Just thought i'd add it so we don't forget this is an issue

BarbourSmith, 2020-05-26:
Right, I know we have talked about this for a while without making any headway. My hunch is that it has to do with trying to iterate through an array while modifying it

alzatin, 2020-06-13:
Did you happen to fix this, it seems to be working perfectly for me now. Are you still seeing this issue?

BarbourSmith, 2020-06-13:
I'm still seeing situations where I can select multiple atoms and pressing delete deletes some but not all of them. The ones which don't delete remain selected and I can delete them by pressing delete a second time

BarbourSmith, 2020-06-13:
I think a solution that would solve this problem would be to delete each atom one at a time something like:

```
selectedAtomsList.forEach(selectedAtom => {
    currentMolecule.nodesOnTheScreen.forEach(nodeOnTheScreen => {
        if(nodeOnTheScreen == selectedAtom){
            nodeOnTheScreen.deleteSelf()
        }
    }
}
```

That would delete each one one after the other in a way which is independent so if one throws an error or something the process won't stop

PR #456 (2020-06-25): Delete multiple atoms fix
Fixes #428 

This should make the delete work through an array of selected atoms to be deleted instead of the delete being activated by keypress in each individual atom.
closed 2020-06-25 by BarbourSmith