MaslowCNC/Maslow-Create #435: Control + S after pressing the "up one level button" doesn't save

BarbourSmith, 2020-05-28:
**Describe the bug**
I'm loving the new control+S feature, but I noticed that using it right after pressing the "Go up one level" button doesn't work

**To Reproduce**
Steps to reproduce the behavior:
1. Click go up one level
2. Press control+S

**Expected behavior**
Saving

**Desktop (please complete the following information):**
 - Windows/Firefox

alzatin, 2020-05-28:
so.. if i just take out the requirement to have an active main body element it saves just fine. The problem is that after pressing any menu button the button remains an active element and I'm not sure how to put the focus back to the mainbody since calling a background click doesn't seem to do it   :S
[image]

BarbourSmith, 2020-05-29:
I think a background click isn't what we want because that's our code to handle someone clicking on the background so it doesn't change focus. I think you want to intentionally give focus using something like this:

https://developer.mozilla.org/en-US/docs/Web/API/HTMLOrForeignElement/focus

PR #451 (2020-06-13): Focus element
Fixes #435 

Puts focus back on mainBody after clicking a button
closed 2020-06-13 by BarbourSmith