MaslowCNC/Maslow-Create #460: Name of inputs fails to update in sidebar

alzatin, 2020-06-24:
**Describe the bug**
When multiple inputs are created the value of the sidebar fails to update name changes to each input

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
When name of input is updated global value names should also update

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

BarbourSmith, 2020-06-25:
I'm not sure I understand what is happening. Can you put in the steps to make it so I can go through them?

alzatin, 2020-06-25:
Yeah sorry I just saw it once and hadn't figured out how to do it again but I didn't want to forget. 
-In a project create 3+ inputs one after the other
- Click on each one and change their names 
- background click 
- Names of the inputs in the global sidebar haven't changed or reflect the name of the first input.

BarbourSmith, 2020-06-25:
Gotcha. A bug for sure.

PR #475 (2020-07-17): Name input update
Fixes #460 

I think the problem with the input name update on side bar had to do with the name is was given by the shortcut when it was creating atoms and i think this fixes it.

Fixes #458 

Recalls set values from io values before promise.all is returned.
closed 2020-07-17 by BarbourSmith