MaslowCNC/Maslow-Create #471: Loading a new project while saving overwrite issue

BarbourSmith, 2020-07-17:
**Describe the bug**

Right now if you open a new project while an old project is saving the old project saves into the new one which was just opened erasing it! This is a big problem because you can accidentally overwrite whole projects.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a complex project which takes a while to save
2. Press save
3. Quickly, before the save finishes open a different project
4. See that the newly open project has been overwritten by the previous project

**Expected behavior**
Ideally the old project would continue saving to the correct place. If that isn't possible then I think the save should be canceled when a new project is opened.

alzatin, 2020-07-26:
Or maybe if while it's saving someone tries to open a new project you could give a warning that saving will stop if you continue or something like that?

alzatin, 2020-07-26:
does it happen only if it's in the same window? or if you are working on projects on different tabs it also happens ?

alzatin, 2020-07-29:
I think once the save function is in motion we can't stop it but we could do a pop up or something that makes the menu wait until the save finishes and tells the person that the menu will open once the project has finished saving.

BarbourSmith, 2020-07-29:
What if right before we do the final push to github we check to see if the project being saved has the same name as the open project or something? That way we wouldn't have to stop the save in progress externally when a new project is loaded, but we could still prevent it from writing to the wrong place

closed 2020-10-05 by github-actions[bot]